tockn/MPU6050_tockn #1: Angle X Y does not work

palm369, 2018-03-29:
Hi, in you example sketches the angle X and Y do not update and are always zero. Only Z updates.

wave100, 2018-04-09:
Running into the same issue here. Anybody know of a fix?

odettman, 2018-04-10:
I'm seeing the same thing. What about an alternative library?

eitanwass, 2018-04-10:
Traced the issue back to the .cpp file where the default values of 0.98f and 0.02f are not passed as acceleration coefficient and gyro coefficient (respectively).
Will update here if I solve the problem.

eitanwass, 2018-04-10:
FIX:
open the cpp and the h file and change the line accordingly:
on CPP:
`MPU6050::MPU6050(TwoWire &w, float accCoeff, float gyroCoeff){
	wire = &w;
	
	accCoef = accCoeff;
	gyroCoef = gyroCoeff;
}`

O the H:
`MPU6050(TwoWire &w, float accCoeff = 0.98f, float gyroCoeff = 0.02f);`

Tockn, 2018-04-10:
fixed it.  
Thanks a lot palm369 and eitanwass!

closed 2018-04-10 by Tockn