tockn/MPU6050_tockn #2: More than one MPU6050

balepari, 2018-04-26:
Is it possible to communicate with more than one MPU6050 on same bus? (i.e. with 2 MPU6050, the 1st has address 0x68, the 2nd has pin AD0 at high level so its address become 0x69).
How is it possible to handle this?

rtek1000, 2018-05-22:
Hello,
We can use the "Auxiliary I2C Bus":
(Pag 26: https://www.invensense.com/wp-content/uploads/2015/02/MPU-6000-Datasheet1.pdf)

> Auxiliary I2C Bus Modes of Operation:
> --> I2C Master Mode: Allows the MPU-60X0 to directly access the data registers of external digital sensors, such as a magnetometer. In this mode, the MPU-60X0 directly obtains data from auxiliary sensors, allowing the on-chip DMP to generate sensor fusion data without intervention from the system applications processor.
> For example, In I2C Master mode, the MPU-60X0 can be configured to perform burst reads, returning the following data from a magnetometer:
> --> X magnetometer data (2 bytes)
> --> Y magnetometer data (2 bytes)
> --> Z magnetometer data (2 bytes)
> The I2C Master can be configured to read up to 24 bytes from up to 4 auxiliary sensors. A fifth sensor can be configured to work single byte read/write mode.
> --> Pass-Through Mode: Allows an external system processor to act as master and directly communicate to the external sensors connected to the auxiliary I2C bus pins (AUX_DA and AUX_CL). In this mode, the auxiliary I2C bus control logic (3rd party sensor interface block) of the MPU-60X0 is disabled, and the auxiliary I2C pins AUX_DA and AUX_CL (Pins 6 and 7) are connected to the main I2C bus (Pins 23 and 24) through analog switches.
> Pass-Through Mode is useful for configuring the external sensors, or for keeping the MPU-60X0 in a low-power mode when only the external sensors are used.
> In Pass-Through Mode the system processor can still access MPU-60X0 data through the I2C interface.

aniketdhole07, 2019-07-04:
Attach second MPU's AD0 to 3.3v, and change the address in the library to 0x69.
Or use my edited library.
https://drive.google.com/open?id=1juFZY0hyuCkK9u3ZHXYO6VHQyksvaFEe

code:
https://drive.google.com/open?id=1uSsEOAKMmNqek6pgB7nI05FxAC8tPLCR

pjphel, 2020-01-18:
Aniketdhole07 -   thanks for your work on this.  I attempted to use your library and into, but could not compile.  The MPU6050_tockn69.h file was identical to the MPU6050_tockn.h file.  I assumed that references to MPU6050 in that file should be MPU605069, so changed each, but it did not compile either.  what am I missing?  thanks!

aniketdhole07, 2020-01-18:
@pjphe Extract the Edited Library and just paste in Arduino's Library folder along with original tockn library.
And try to complie.
Dont just add .h file,add complete folder,I have changed the Address to 69H and Class name to MPU60502

pjphel, 2020-01-18:
I replaced each reference to MPU6050 with MPU605069 in MPU6050_tockn69.h and it works flawlessly.  I had missed a reference or two previously.  Thanks again!!, this is a hugely useful tool to have.   Seems like the file you uploaded needs updating.

darekop, 2020-01-18:
Is this what MPU6050_tockn69.h file looks like after changes?

```
#ifndef MPU60506969_TOCKN_H
#define MPU605069_TOCKN_H

#include "Arduino.h"
#include "Wire.h"

#define MPU605069_ADDR         0x69
#define MPU605069_SMPLRT_DIV   0x19
#define MPU605069_CONFIG       0x1a
#define MPU605069_GYRO_CONFIG  0x1b
#define MPU605069_ACCEL_CONFIG 0x1c
#define MPU605069_WHO_AM_I     0x75
#define MPU605069_PWR_MGMT_1   0x6b
#define MPU605069_TEMP_H       0x41
#define MPU605069_TEMP_L       0x42

class MPU605069{
  public:

  MPU605069(TwoWire &w);
  MPU605069(TwoWire &w, float aC, float gC);

  void begin();

  void setGyroOffsets(float x, float y, float z);

  void writeMPU605069(byte reg, byte data);
  byte readMPU605069(byte reg);

  int16_t getRawAccX(){ return rawAccX; };
  int16_t getRawAccY(){ return rawAccY; };
  int16_t getRawAccZ(){ return rawAccZ; };

  int16_t getRawTemp(){ return rawTemp; };

  int16_t getRawGyroX(){ return rawGyroX; };
  int16_t getRawGyroY(){ return rawGyroY; };
  int16_t getRawGyroZ(){ return rawGyroZ; };

  float getTemp(){ return temp; };

  float getAccX(){ return accX; };
  float getAccY(){ return accY; };
  float getAccZ(){ return accZ; };

  float getGyroX(){ return gyroX; };
  float getGyroY(){ return gyroY; };
  float getGyroZ(){ return gyroZ; };

	void calcGyroOffsets(bool console = false, uint16_t delayBefore = 1000, uint16_t delayAfter = 3000);

  float getGyroXoffset(){ return gyroXoffset; };
  float getGyroYoffset(){ return gyroYoffset; };
  float getGyroZoffset(){ return gyroZoffset; };

  void update();

  float getAccAngleX(){ return angleAccX; };
  float getAccAngleY(){ return angleAccY; };

  float getGyroAngleX(){ return angleGyroX; };
  float getGyroAngleY(){ return angleGyroY; };
  float getGyroAngleZ(){ return angleGyroZ; };

  float getAngleX(){ return angleX; };
  float getAngleY(){ return angleY; };
  float getAngleZ(){ return angleZ; };

  private:

  TwoWire *wire;

  int16_t rawAccX, rawAccY, rawAccZ, rawTemp,
  rawGyroX, rawGyroY, rawGyroZ;

  float gyroXoffset, gyroYoffset, gyroZoffset;

  float temp, accX, accY, accZ, gyroX, gyroY, gyroZ;

  float angleGyroX, angleGyroY, angleGyroZ,
  angleAccX, angleAccY, angleAccZ;

  float angleX, angleY, angleZ;

  float interval;
  long preInterval;

  float accCoef, gyroCoef;
};

#endif
```
The changed file does not compile

sorry for my English

pjphel, 2020-01-18:
yes except the top two lines are: 

#ifndef MPU605069_TOCKN_H
#define MPU605069_TOCKN_H

I think you had one extra 69 in there.

darekop, 2020-01-18:
the MPU6050_tockn69.h file looks like this

```
#ifndef MPU6050_TOCKN_H
#define MPU6050_TOCKN_H

#include "Arduino.h"
#include "Wire.h"

#define MPU605069_ADDR         0x69
#define MPU605069_SMPLRT_DIV   0x19
#define MPU605069_CONFIG       0x1a
#define MPU605069_GYRO_CONFIG  0x1b
#define MPU605069_ACCEL_CONFIG 0x1c
#define MPU605069_WHO_AM_I     0x75
#define MPU605069_PWR_MGMT_1   0x6b
#define MPU605069_TEMP_H       0x41
#define MPU605069_TEMP_L       0x42

class MPU605069{
  public:

  MPU605069(TwoWire &w);
  MPU605069(TwoWire &w, float aC, float gC);

  void begin();

  void setGyroOffsets(float x, float y, float z);

  void writeMPU605069(byte reg, byte data);
  byte readMPU605069(byte reg);

  int16_t getRawAccX(){ return rawAccX; };
  int16_t getRawAccY(){ return rawAccY; };
  int16_t getRawAccZ(){ return rawAccZ; };

  int16_t getRawTemp(){ return rawTemp; };

  int16_t getRawGyroX(){ return rawGyroX; };
  int16_t getRawGyroY(){ return rawGyroY; };
  int16_t getRawGyroZ(){ return rawGyroZ; };

  float getTemp(){ return temp; };

  float getAccX(){ return accX; };
  float getAccY(){ return accY; };
  float getAccZ(){ return accZ; };

  float getGyroX(){ return gyroX; };
  float getGyroY(){ return gyroY; };
  float getGyroZ(){ return gyroZ; };

	void calcGyroOffsets(bool console = false, uint16_t delayBefore = 1000, uint16_t delayAfter = 3000);

  float getGyroXoffset(){ return gyroXoffset; };
  float getGyroYoffset(){ return gyroYoffset; };
  float getGyroZoffset(){ return gyroZoffset; };

  void update();

  float getAccAngleX(){ return angleAccX; };
  float getAccAngleY(){ return angleAccY; };

  float getGyroAngleX(){ return angleGyroX; };
  float getGyroAngleY(){ return angleGyroY; };
  float getGyroAngleZ(){ return angleGyroZ; };

  float getAngleX(){ return angleX; };
  float getAngleY(){ return angleY; };
  float getAngleZ(){ return angleZ; };

  private:

  TwoWire *wire;

  int16_t rawAccX, rawAccY, rawAccZ, rawTemp,
  rawGyroX, rawGyroY, rawGyroZ;

  float gyroXoffset, gyroYoffset, gyroZoffset;

  float temp, accX, accY, accZ, gyroX, gyroY, gyroZ;

  float angleGyroX, angleGyroY, angleGyroZ,
  angleAccX, angleAccY, angleAccZ;

  float angleX, angleY, angleZ;

  float interval;
  long preInterval;

  float accCoef, gyroCoef;
};
erial.print(mpu60502.getAngleY()-y2);
  Serial.print("\tangleZ ****** : ");
  Serial.println
#endif

```
My sketch:
```
#include<MPU6050_tockn.h>
#include <MPU6050_tockn69.h>
#include <Wire.h>
int x1,x2,y1,y2,z1,z2;

MPU605069 mpu60502(Wire);
MPU6050 mpu6050(Wire);
void setup() {
  Serial.begin(9600);
  Wire.begin();
  mpu60502.begin();
  mpu60502.calcGyroOffsets(true);
  mpu6050.begin();
  mpu6050.calcGyroOffsets(true);
  mpu60502.update();
  x2=mpu60502.getAngleX();
   y2=mpu60502.getAngleY();
    z2=mpu60502.getAngleZ();
    mpu6050.update();
     x1=mpu6050.getAngleX();
      y1=mpu6050.getAngleY();
       z1=mpu6050.getAngleZ();
}

void loop() {
  mpu60502.update();//mpu2
  Serial.print("angleX ****** : "); //x2=z1
  Serial.print(mpu60502.getAngleX()-x2);
  Serial.print("\tangleY******* : ");
  S(mpu60502.getAngleZ()-z2);
  
  mpu6050.update();//mpu1
  Serial.print("angleX : ");
  Serial.print(mpu6050.getAngleX()-x1);//x1=z2
  Serial.print("\tangleY : ");
  Serial.print(mpu6050.getAngleY()-y1);
  Serial.print("\tangleZ : ");
  Serial.println(mpu6050.getAngleZ()-z1);
 
}
```

compilation effect. sketch - test69.ino

> test69:8:1: error: 'MPU605069' does not name a type; did you mean 'MPU6050'?
> 
>  MPU605069 mpu60502(Wire);
> 
>  ^~~~~~~~~
> 
>  MPU6050
> 
> C:\Users\Darek\Downloads\test69\test69.ino: In function 'void setup()':
> 
> test69:13:3: error: 'mpu60502' was not declared in this scope
> 
>    mpu60502.begin();
> 
>    ^~~~~~~~
> 
> C:\Users\Darek\Downloads\test69\test69.ino:13:3: note: suggested alternative: 'mpu6050'
> 
>    mpu60502.begin();
> 
>    ^~~~~~~~
> 
>    mpu6050
> 
> C:\Users\Darek\Downloads\test69\test69.ino: In function 'void loop()':
> 
> test69:28:3: error: 'mpu60502' was not declared in this scope
> 
>    mpu60502.update();//mpu2
> 
>    ^~~~~~~~
> 
> C:\Users\Darek\Downloads\test69\test69.ino:28:3: note: suggested alternative: 'mpu6050'
> 
>    mpu60502.update();//mpu2
> 
>    ^~~~~~~~
> 
>    mpu6050
> 
> Znaleziono wiele bibliotek w "MPU6050_tockn.h"
> Wykorzystane: C:\Users\Darek\Documents\Arduino\libraries\MPU6050_tockn-master
> Znaleziono wiele bibliotek w "Wire.h"
> Wykorzystane: C:\Program
> Znaleziono wiele bibliotek w "MPU6050_tockn69.h"
> Wykorzystane: C:\Users\Darek\Documents\Arduino\libraries\MPU6050_tockn-master69
> exit status 1
> 'MPU605069' does not name a type; did you mean 'MPU6050'?

aniketdhole07, 2020-01-18:
You need to change function names in .cpp file also

darekop, 2020-01-18:
File - MPU6050_tockn69.h

```
#ifndef MPU6050_TOCKN_H
#define MPU6050_TOCKN_H

#include "Arduino.h"
#include "Wire.h"

#define MPU605069_ADDR         0x69
#define MPU605069_SMPLRT_DIV   0x19
#define MPU605069_CONFIG       0x1a
#define MPU605069_GYRO_CONFIG  0x1b
#define MPU605069_ACCEL_CONFIG 0x1c
#define MPU605069_WHO_AM_I     0x75
#define MPU605069_PWR_MGMT_1   0x6b
#define MPU605069_TEMP_H       0x41
#define MPU605069_TEMP_L       0x42

class MPU605069{
  public:

  MPU605069(TwoWire &w);
  MPU605069(TwoWire &w, float aC, float gC);

  void begin();

  void setGyroOffsets(float x, float y, float z);

  void writeMPU605069(byte reg, byte data);
  byte readMPU605069(byte reg);

  int16_t getRawAccX(){ return rawAccX; };
  int16_t getRawAccY(){ return rawAccY; };
  int16_t getRawAccZ(){ return rawAccZ; };

  int16_t getRawTemp(){ return rawTemp; };

  int16_t getRawGyroX(){ return rawGyroX; };
  int16_t getRawGyroY(){ return rawGyroY; };
  int16_t getRawGyroZ(){ return rawGyroZ; };

  float getTemp(){ return temp; };

  float getAccX(){ return accX; };
  float getAccY(){ return accY; };
  float getAccZ(){ return accZ; };

  float getGyroX(){ return gyroX; };
  float getGyroY(){ return gyroY; };
  float getGyroZ(){ return gyroZ; };

	void calcGyroOffsets(bool console = false, uint16_t delayBefore = 1000, uint16_t delayAfter = 3000);

  float getGyroXoffset(){ return gyroXoffset; };
  float getGyroYoffset(){ return gyroYoffset; };
  float getGyroZoffset(){ return gyroZoffset; };

  void update();

  float getAccAngleX(){ return angleAccX; };
  float getAccAngleY(){ return angleAccY; };

  float getGyroAngleX(){ return angleGyroX; };
  float getGyroAngleY(){ return angleGyroY; };
  float getGyroAngleZ(){ return angleGyroZ; };

  float getAngleX(){ return angleX; };
  float getAngleY(){ return angleY; };
  float getAngleZ(){ return angleZ; };
[... 147 lines cut ...]

  gyroX = ((float)rawGyroX) / 65.5;
  gyroY = ((float)rawGyroY) / 65.5;
  gyroZ = ((float)rawGyroZ) / 65.5;

  gyroX -= gyroXoffset;
  gyroY -= gyroYoffset;
  gyroZ -= gyroZoffset;

  interval = (millis() - preInterval) * 0.001;

  angleGyroX += gyroX * interval;
  angleGyroY += gyroY * interval;
  angleGyroZ += gyroZ * interval;

  angleX = (gyroCoef * (angleX + gyroX * interval)) + (accCoef * angleAccX);
  angleY = (gyroCoef * (angleY + gyroY * interval)) + (accCoef * angleAccY);
  angleZ = angleGyroZ;

  preInterval = millis();

}

```
The program does not compile

darekop, 2020-01-19:
Unfortunately I can't deal with this library. 
Can I ask you to insert or send [darekopop@gmail.com](url) correctly changed files?

Thanks!

pjphel, 2020-01-23:
MPU6050_tockn69.cpp

#include "MPU6050_tockn69.h"
#include "Arduino.h"

MPU605069::MPU605069(TwoWire &w){
  wire = &w;
  accCoef = 0.02f;
  gyroCoef = 0.98f;
}

MPU605069::MPU605069(TwoWire &w, float aC, float gC){
  wire = &w;
  accCoef = aC;
  gyroCoef = gC;
}

void MPU605069::begin(){
  writeMPU605069(MPU605069_SMPLRT_DIV, 0x00);
  writeMPU605069(MPU605069_CONFIG, 0x00);
  writeMPU605069(MPU605069_GYRO_CONFIG, 0x08);
  writeMPU605069(MPU605069_ACCEL_CONFIG, 0x00);
  writeMPU605069(MPU605069_PWR_MGMT_1, 0x01);
  this->update();
  angleGyroX = 0;
  angleGyroY = 0;
  angleX = this->getAccAngleX();
  angleY = this->getAccAngleY();
  preInterval = millis();
}

void MPU605069::writeMPU605069(byte reg, byte data){
  wire->beginTransmission(MPU605069_ADDR);
  wire->write(reg);
  wire->write(data);
  wire->endTransmission();
}

byte MPU605069::readMPU605069(byte reg) {
  wire->beginTransmission(MPU605069_ADDR);
  wire->write(reg);
  wire->endTransmission(true);
  wire->requestFrom(MPU605069_ADDR, 1);
  byte data =  wire->read();
  return data;
}

void MPU605069::setGyroOffsets(float x, float y, float z){
  gyroXoffset = x;
  gyroYoffset = y;
  gyroZoffset = z;
}

void MPU605069::calcGyroOffsets(bool console, uint16_t delayBefore, uint16_t delayAfter){
	float x = 0, y = 0, z = 0;
	int16_t rx, ry, rz;

  delay(delayBefore);
	if(console){
    Serial.println();
    Serial.println("========================================");
    Serial.println("Calculating gyro offsets");
    Serial.print("DO NOT MOVE MPU6050");
  }
  for(int i = 0; i < 3000; i++){
    if(console && i % 1000 == 0){
      Serial.print(".");
    }
    wire->beginTransmission(MPU605069_ADDR);
    wire->write(0x43);
[... 141 lines cut ...]
  float getAngleZ(){ return angleZ; };

  private:

  TwoWire *wire;

  int16_t rawAccX, rawAccY, rawAccZ, rawTemp,
  rawGyroX, rawGyroY, rawGyroZ;

  float gyroXoffset, gyroYoffset, gyroZoffset;

  float temp, accX, accY, accZ, gyroX, gyroY, gyroZ;

  float angleGyroX, angleGyroY, angleGyroZ,
  angleAccX, angleAccY, angleAccZ;

  float angleX, angleY, angleZ;

  float interval;
  long preInterval;

  float accCoef, gyroCoef;
};

#endif

darekop, 2020-01-23:
Everything is OK.

Thank you very much

netpipe, 2021-11-20:
https://github.com/netpipe/HappyHands/tree/main/gyroArduino/MPU6050_tockn-swire

darekop, 2022-05-11:
This is my code:
```
#include <LiquidCrystal.h>
#include <MPU6050_tockn.h>
#include <MPU6050_tockn69.h>
#include <Wire.h>
#include <SPI.h>

LiquidCrystal lcd(12, 11, 10, 9, 8, 7);

MPU6050 mpu6050(Wire);
MPU605069 mpu60502(Wire);
float wynikx1 = 0;
float wynikx2 = 0;
float x1, x2, y1, y2, z1, z2;

void setup() {
  Serial.begin(9600);

  Wire.begin();
  mpu6050.begin();
  mpu6050.calcGyroOffsets(true);
  mpu60502.begin();
  mpu60502.calcGyroOffsets(true);
  

  
  lcd.begin(20, 2);
  
  lcd.setCursor(0, 0);
  lcd.print("START");

  delay(1000);
  lcd.setCursor(0, 0);
  lcd.print("     ");

  x1=-1.9;
  y1=-3.3;
  x2=1.5;
  y2=1;
}

void loop() {
  
  lcd.setCursor(0, 0);
  lcd.print("x1=");
  lcd.setCursor(10, 0);
  lcd.print("y1=");
  lcd.setCursor(0, 1);
  lcd.print("x2=");
  lcd.setCursor(10, 1);
  lcd.print("y2=");

    

  mpu6050.update();
  lcd.setCursor(3, 0);
  lcd.print((mpu6050.getAngleX() - x1), 1);
  lcd.print("  ");
  lcd.setCursor(13, 0);
  lcd.print((mpu6050.getAngleY() - y1), 1);
  lcd.print("  ");

  mpu60502.update();
  lcd.setCursor(3, 1);
  lcd.print((mpu60502.getAngleX() - x2), 1);
  lcd.print("  ");
lcd.setCursor(13, 1);
  lcd.print((mpu60502.getAngleY() - y2), 1);
  lcd.print("  ");

}
```

I reset the indications by setting both sensors on a flat surface
[image]

Then I tilt both sensors to 30, 45 and 90 degrees
[image]

[image]

[image]

Unfortunately, the results are different for both sensors. I swapped sensors, replaced them with others, and the indications were never the same.

HELP I NO LONGER HAVE IDEAS