tockn/MPU6050_tockn #4: Change Accelerometer and gyroscope sensitivity

Smartheads, 2018-05-21:
Please add new methods to be able to change the sensors sensitivity settings.

Information on MPU6050 sensitivity settings: [https://www.invensense.com/products/motion-tracking/6-axis/mpu-6050/](url)

Thanks

rtek1000, 2018-05-22:
Hello,

Let's go to look at MPU6050_tockn.h file (/src/MPU6050_tockn.h path) the address below are of registers found at Register Map doc we can find here:
https://www.invensense.com/wp-content/uploads/2015/02/MPU-6000-Register-Map1.pdf

Register Map - Pag 14: 4.4 Register 27 – Gyroscope Configuration
MPU6050_tockn.h - line 10: #define MPU6050_GYRO_CONFIG  0x1b

Register Map - Pag 15: 4.5 Register 28 – Accelerometer Configuration
MPU6050_tockn.h - line 11: #define MPU6050_ACCEL_CONFIG 0x1c

These addr are used in MPU6050_tockn.cpp file:
Line 19: writeMPU6050(MPU6050_GYRO_CONFIG, 0x08);
Line 20: writeMPU6050(MPU6050_ACCEL_CONFIG, 0x00);

Look at line 19, 0x08 is a hex number, or 0B1000 (binary) this will set Bit3 of GYRO_CONFIG register as High state, or FS_SEL with value 1.
Register Map - Pag 14: FS_SEL = 1 ==> Full Scale Range = ± 500 °/s

Datasheet show the [PARAMETER] "Sensitivity Scale Factor" at page 12:
[CONDITIONS] FS_SEL=1 [TYP] 65.5 [UNITS] LSB/(º/s)

The MPU6050_tockn.cpp file contains 65.5 in lines 78, 79 and 80:
		x += ((float)rx) / 65.5;
		y += ((float)ry) / 65.5;
		z += ((float)rz) / 65.5;

And lines 119, 120 and 121:
	gyroX = ((float)rawGyroX) / 65.5;
	gyroY = ((float)rawGyroY) / 65.5;
	gyroZ = ((float)rawGyroZ) / 65.5;

Based on the gyroscope, we can check the accelerometer configuration ;)

Of course implementing FS_SEL methods will help.

Smartheads, 2018-07-07:
Thanks for your help rtek1000!

Finally got down to resolving the issue. Created new pull request. (Sorry to keep anybody waiting, if they were)

I'll close the issue, when my pull request is accepted.

jjarp, 2020-05-28:
Hello thanks a lot for the explanation and this good library.👍

I already changed the line 19 with 0x18 (for 2000°/s)
Line 19: writeMPU6050(MPU6050_GYRO_CONFIG, 0x18);

but I can not understand this part: 
**_Datasheet show the [PARAMETER] "Sensitivity Scale Factor" at page 12:
[CONDITIONS] FS_SEL=1 [TYP] 65.5 [UNITS] LSB/(º/s)_**

I am trying to find it in the datasheet but I can't, may be can you explain me with an example?

thanks a lot in advance.

Smartheads, 2020-06-01:
Hi!

So what I think rtek ment, was that if you set FS_SEL to 1, then the **sensitivity scale factor** will be 65.5 LSB/deg/s.

The MPU6050 gyroscope has a data resolution of 65500 units, which means that the MPU6050 will produce a reading between -32750 and +32750. This value then has to be scaled down based on the **sensitivity scale factor** to get a reading in deg/s.

Here's a table containing the scale factors for each sensitivity range:
[image]

Otherwise this can be calculated by simple dividing the data resolution by the size of the range. (for +-250 deg/s -> size of range: 500 -> 65500 / 500 = 131)

So when you right your code, be sure to select the right scale factor on lines 78, 79, 80 and 119, 120, 121.

Example if FS_SEL is set to 2 (+-1000 deg/s):

```
x += ((float)rx) / 32.8f;
y += ((float)ry) / 32.8f;
z += ((float)rz) / 32.8f;
```

Hope I was able to answer your question.

rtek1000, 2020-06-01:
Hello,

The datasheet can be found this way:

[image]