tockn/MPU6050_tockn #9: Drifting on MPU6050

LuckyJMD, 2018-08-07:
Hello,
great job, works nice. sorry if my issue is unrelated but I am trying to make a self-balancing robot and I am getting drifting on gyro angle like it is initially 0.01 degree and after a  couple back and forth movements the gyro angle becomes 0.20 in stationary position.. is this fixable or totally my own issue?

thank you for your time!

note: in README.md exaple codes, the #include <MPU6050_tockn> does not include .h so I got issues while uploading. if you can change them to #include <MPU6050_tockn.h> would be better :)

Smartheads, 2018-08-09:
Hi!
Yea, drift is a common problem with accelerometers and gyroscopes. What you can do to minimise it is to:
* calibrate the accelerometer and gyroscope
* change the sensitivity settings to be less sensitive
* write an algorithm to return the average or median of the gyro readings and implement your own mpu6050.update()

I hope these tips help!

(Btw there might be a better equipt library for the mpu6050. I myself am working on one, but its still under development (it has methods, like getAccelXMedian()...)

silentwarrior, 2019-10-25:
Did anyone find a solution for this?

I keep on getting drift and makes it quite unusable.

Some axis increase gyro angle by almost 1 angle per second :/