tockn/MPU6050_tockn #15: Angle Inaccuracy - Please Advise

sparfitt93, 2018-10-25:
Hi,
First thanks fro the code, it's by far the easiest to use for the MPU, but I have noticed the angles in X and Y are consistently off. 

I made a 3D printed block with surfaces at 0, 30, 45, 60 and 90 degrees to test and found that both X and Y have issues. 
At 0 degrees the error is 1-2 degrees (Not an issue)
However, at 30 degrees both X and Y read between 23-26,
at 45 both read around 34-37,
at 60 both read 46-48 
and at 90 both read about 75.

So about 15-25% error.

Any reason for this? I tried varying the accel/gyro ratio and nothing seemed to effect it.

Thanks in advance.

sparfitt93, 2018-10-25:
Update - Increasing the values output but 20% gives much better results, only 1-2 degrees of variation.

wearni, 2018-11-06:
Hi. What did you actually do to increase output values 20%?

sparfitt93, 2018-11-06:
Literally multiplied the output by a factor of 1.2 

`  y_angle = 1.2*mpu.getAngleY();`

dzulkiflijoey, 2019-03-20:
May i know what the is the unit for the reading that have been shown by this coding? in degree degrees per second or in radian  per second?