tockn/MPU6050_tockn #16: I2C Read Issue

MichiTheSailor, 2018-12-22:
Ciao,

First o fall many thanks for your good work - Make handling of MPU6050 really easy ;o)

After Updating to new version of ESP32-HAL-I2C I realized that READ from MPU6050 did not work any more. (got only NULL values) 
After research and comparision with different other includes like I2C-EEPROM, I2C-Display, etc. I realized that your access seems wrong designed and worked until now only by accident.

Real Bugfix is the wrong parameter in "endTransmission", further I added e rought "error detection/handling".

Simple Bugfix in MPU6050_tockn.cpp:
Change Line 71 in MPU6050::calcGyroOffsets from: 
      wire->endTransmission(false)
to:
	**byte rxStatus =** wire->endTransmission(**true**);
	**if (rxStatus != 0) Serial.println("MPU6050::update EC: " + String(rxStatus));**

Same Bugfix in MPU6050_tockn.cpp:
Change Line 105 in MPU6050::update from: 
      wire->endTransmission(false)
to:
	**byte rxStatus =** wire->endTransmission(**true**);
	**if (rxStatus != 0) Serial.println("MPU6050::update EC: " + String(rxStatus));**

Afterwards it worked really fine....  :o)

Kind regards
Michi

Redstoned, 2018-12-31:
This fix worked also fine for me on my ESP32 DOIT Devkit V1.
The Serial.print didn't worked for me. Without it it works.

CaNiRaspUsomethinG, 2019-01-08:
Still not working for me on my ESP32 Doit Devkit...

agunes77, 2019-02-17:
thank you Damian

netpipe, 2021-11-20:
https://github.com/netpipe/HappyHands/tree/main/gyroArduino/MPU6050_tockn-swire