tockn/MPU6050_tockn #18: Compilation Issue

neilbalch, 2019-03-08:
Upon compilation:

```
Compiling .pioenvs\genericSTM32F103C8\lib0d0\MPU6050_tockn_ID2824\MPU6050_tockn.cpp.o
C:\Users\neilb\.platformio\lib\MPU6050_tockn_ID2824\src\MPU6050_tockn.cpp: In member function 'void MPU6050::calcGyroOffsets(bool)':
C:\Users\neilb\.platformio\lib\MPU6050_tockn_ID2824\src\MPU6050_tockn.cpp:71:53: error: no matching function for call to 'TwoWire::requestFrom(int, int, int)'
wire->requestFrom((int)MPU6050_ADDR, 14, (int)true);
^
In file included from C:\Users\neilb\.platformio\packages\framework-arduinoststm32-maple\STM32F1\libraries\Wire/Wire.h:42:0,
from C:\Users\neilb\.platformio\lib\MPU6050_tockn_ID2824\src\MPU6050_tockn.h:5,
from C:\Users\neilb\.platformio\lib\MPU6050_tockn_ID2824\src\MPU6050_tockn.cpp:1:
C:\Users\neilb\.platformio\packages\framework-arduinoststm32-maple\STM32F1\libraries\Wire/utility/WireBase.h:101:11: note: candidate: uint8 WireBase::requestFrom(uint8, int)
uint8 requestFrom(uint8, int);
^~~~~~~~~~~
C:\Users\neilb\.platformio\packages\framework-arduinoststm32-maple\STM32F1\libraries\Wire/utility/WireBase.h:101:11: note:   candidate expects 2 arguments, 3 provided
C:\Users\neilb\.platformio\packages\framework-arduinoststm32-maple\STM32F1\libraries\Wire/utility/WireBase.h:106:11: note: candidate: uint8 WireBase::requestFrom(int, int)
uint8 requestFrom(int, int);
^~~~~~~~~~~
C:\Users\neilb\.platformio\packages\framework-arduinoststm32-maple\STM32F1\libraries\Wire/utility/WireBase.h:106:11: note:   candidate expects 2 arguments, 3 provided
C:\Users\neilb\.platformio\lib\MPU6050_tockn_ID2824\src\MPU6050_tockn.cpp:73:21: warning: value computed is not used [-Wunused-value]
wire->read() << 8 | wire->read();
~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~
C:\Users\neilb\.platformio\lib\MPU6050_tockn_ID2824\src\MPU6050_tockn.cpp:74:21: warning: value computed is not used [-Wunused-value]
wire->read() << 8 | wire->read();
~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~
C:\Users\neilb\.platformio\lib\MPU6050_tockn_ID2824\src\MPU6050_tockn.cpp:75:21: warning: value computed is not used [-Wunused-value]
wire->read() << 8 | wire->read();
~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~
C:\Users\neilb\.platformio\lib\MPU6050_tockn_ID2824\src\MPU6050_tockn.cpp:76:21: warning: value computed is not used [-Wunused-value]
wire->read() << 8 | wire->read();
~~~~~~~~~~~~~~~~~~^~~~~~~~~~~~~~
C:\Users\neilb\.platformio\lib\MPU6050_tockn_ID2824\src\MPU6050_tockn.cpp: In member function 'void MPU6050::update()':
C:\Users\neilb\.platformio\lib\MPU6050_tockn_ID2824\src\MPU6050_tockn.cpp:105:52: error: no matching function for call to 'TwoWire::requestFrom(int, int, int)'
wire->requestFrom((int)MPU6050_ADDR, 14, (int)true);
^
In file included from C:\Users\neilb\.platformio\packages\framework-arduinoststm32-maple\STM32F1\libraries\Wire/Wire.h:42:0,
from C:\Users\neilb\.platformio\lib\MPU6050_tockn_ID2824\src\MPU6050_tockn.h:5,
from C:\Users\neilb\.platformio\lib\MPU6050_tockn_ID2824\src\MPU6050_tockn.cpp:1:
C:\Users\neilb\.platformio\packages\framework-arduinoststm32-maple\STM32F1\libraries\Wire/utility/WireBase.h:101:11: note: candidate: uint8 WireBase::requestFrom(uint8, int)
uint8 requestFrom(uint8, int);
^~~~~~~~~~~
C:\Users\neilb\.platformio\packages\framework-arduinoststm32-maple\STM32F1\libraries\Wire/utility/WireBase.h:101:11: note:   candidate expects 2 arguments, 3 provided
C:\Users\neilb\.platformio\packages\framework-arduinoststm32-maple\STM32F1\libraries\Wire/utility/WireBase.h:106:11: note: candidate: uint8 WireBase::requestFrom(int, int)
uint8 requestFrom(int, int);
^~~~~~~~~~~
C:\Users\neilb\.platformio\packages\framework-arduinoststm32-maple\STM32F1\libraries\Wire/utility/WireBase.h:106:11: note:   candidate expects 2 arguments, 3 provided
*** [.pioenvs\genericSTM32F103C8\lib0d0\MPU6050_tockn_ID2824\MPU6050_tockn.cpp.o] Error 1
```

I'm not entitely sure what's up. I'm using PlatformIO, and building for the STM32F103C Blue Pill.

neilbalch, 2019-03-12:
I now know what's going on: https://community.platformio.org/t/stm32f1-compilation-erring-upon-calling-wire-library/6911/14

tockn, 2019-03-18:
Thanks for your report and PR!
refs: #14 #20

closed 2019-03-18 by tockn