tockn/MPU6050_tockn #25: This Library is compatible with ESP32?

cruizg, 2019-04-11:
This Library is compatible with ESP32?.
Can you sendme a example with ESP32 and conections?

Titibo26, 2019-05-27:
Hi,
It's working fine for me. Just make your setup according to your board : 
"
#include <MPU6050_tockn.h>
#include <Wire.h>

MPU6050 mpu6050(Wire);

void setup() {
  Serial.begin(115200); // I use this bauderate for the esp32
  Wire.begin(21,22); //SDA SDL for esp32
  mpu6050.begin();
  mpu6050.calcGyroOffsets(true);
} "

cruizg, 2019-05-27:
This dont work for me, because i have esp32 oled

tockn, 2019-05-27:
Is this the same of your ESP32 OLED?
https://banggood.com/Wemos-ESP32-OLED-Module-For-Arduino-ESP32-OLED-WiFi-Bluetooth-Dual-ESP-32-ESP-32S-ESP8266-p-1148119.html?cur_warehouse=CN

If so, you can use `25` and `26` pin.
You will connect `25` to `SDA` and `26` to `SCL` and following example may work.

```
#include <MPU6050_tockn.h>
#include <Wire.h>

MPU6050 mpu6050(Wire);

void setup() {
  Serial.begin(115200); // I use this bauderate for the esp32
  Wire.begin(26,25); //SDA SDL for esp32
  mpu6050.begin();
  mpu6050.calcGyroOffsets(true);
}

...

```

cruizg, 2019-05-28:
i was try, because the result is ever:

`angleX : -135.00	angleY : 135.00	angleZ : -0.00
angleX : -135.00	angleY : 135.00	angleZ : -0.00
angleX : -135.00	angleY : 135.00	angleZ : -0.00
angleX : -135.00	angleY : 135.00	angleZ : -0.00
angleX : -135.00	angleY : 135.00	angleZ : -0.00
angleX : -135.00	angleY : 135.00	angleZ : -0.00
angleX : -135.00	angleY : 135.00	angleZ : -0.00
angleX : -135.00	angleY : 135.00	angleZ : -0.00
angleX : -135.00	angleY : 135.00	angleZ : -0.00
angleX : -135.00	angleY : 135.00	angleZ : -0.00
angleX : -135.00	angleY : 135.00	angleZ : -0.00
angleX : -135.00	angleY : 135.00	angleZ : -0.00
angleX : -135.00	angleY : 135.00	angleZ : -0.00
`

oliverjc, 2019-07-19:
If you're using the Wemos ESP32 Lolin, I2C uses pins 5 and 4 for SDA and SCL respectively. So in your setup function, you need to have `Wire.begin(5,4);`

zhuwei, 2020-01-03:
That's great! It's working fine for me.Thx!

Ce345, 2020-01-15:
Have similar problem.
Wire.begin(5,4) didn't work (either).
Repeated resets and loads always the same, whatever movement is made:
=======================================================
temp : 36.50
accX : -0.00	accY : -0.00	accZ : -0.00
gyroX : -0.00	gyroY : -0.00	gyroZ : -0.00
accAngleX : -35.26	accAngleY : 35.26
gyroAngleX : -0.00	gyroAngleY : -0.00	gyroAngleZ : -0.00
angleX : -35.26	angleY : 35.26	angleZ : -0.00
=======================================================
Using GEEKCREIT ESP-F DEVKIT V4 and a WEMOS mini gave exact the same results!
Other basic sketches did give changing values.

Ce345, 2020-01-15:
Of course, the text should be smaller!

Ce345, 2020-01-17:
This  library for the MPU6050 is the third tried, with some or no succes at all. 
After delving (today) into problems elsewhere mentioned with the combo ESP and MPU6050, a suggested addition to the code made it working the right way.
After _Wire.begin();_  added the line: __Wire. setClock ( 400000);
It seems that the MPU6050 starts default with 100.000 Hertz, which is the right way for a Arduino, but not for the (in much aspects faster) ESP.

toosimit, 2021-01-15:
> 
> 
> If you're using the Wemos ESP32 Lolin, I2C uses pins 5 and 4 for SDA and SCL respectively. So in your setup function, you need to have `Wire.begin(5,4);`

I have same problem with this code it works before on Arduino pro micro but on esp8266 the output is 

`angleX : -135.00 angleY : 135.00 angleZ : -0.00
angleX : -135.00 angleY : 135.00 angleZ : -0.00
angleX : -135.00 angleY : 135.00 angleZ : -0.00
angleX : -135.00 angleY : 135.00 angleZ : -0.00
angleX : -135.00 angleY : 135.00 angleZ : -0.00
angleX : -135.00 angleY : 135.00 angleZ : -0.00
angleX : -135.00 angleY : 135.00 angleZ : -0.00
angleX : -135.00 angleY : 135.00 angleZ : -0.00
angleX : -135.00 angleY : 135.00 angleZ : -0.00
angleX : -135.00 angleY : 135.00 angleZ : -0.00`

I changed to Wire.begin(5,4) but isn't work.

toosimit, 2021-01-15:
I changed some part and finally it work with es8266 and Arduino IDE 

```

#include <MPU6050_tockn.h>
#include <Wire.h>

MPU6050 mpu6050(Wire);
int X, Y, Z;
void setup() {
  Serial.begin(9600);
  Wire.begin(4,5,0x68);
  Wire. setClock ( 400000);
  mpu6050.begin();
  mpu6050.calcGyroOffsets(true);
}

void loop() {
  X = mpu6050.getAngleX();
  Y = mpu6050.getAngleY();
  Z = mpu6050.getAngleZ();
  mpu6050.update();
  Serial.print("angleX : ");
  Serial.print(X);
  Serial.print("\tangleY : ");
  Serial.print(Y * (-1));
  Serial.print("\tangleZ : ");
  Serial.println(Z);
}
```