tockn/MPU6050_tockn #26: Arduino Fails to upload because of an issue of the library

ramithuh, 2019-04-16:
As stated https://stackoverflow.com/a/29135050/2715780
in this answer, 3 '!' marks in a row causes the Mega 2560 bootloader to go into Monitor mode from which it can not finish programming. Therefore the code doesn't upload and causes the error
**stk500v2_ReceiveMessage(): timeout** 

I manually located the .cpp file in my arduino library and changed "done!!!" to "done!", now it uploaded

ramithuh, 2019-04-16:
[image]

tockn, 2019-04-16:
oh... Thank you for your report. I didn’t know such a specification :(
I will create a patch to fix this issue.

ramithuh, 2019-04-16:
You're welcome.. Thank you this amazing library! :)

tockn, 2019-04-16:
Thank you for your PR!
I will merge that!

closed 2019-04-17 by ramithuh