tockn/MPU6050_tockn #33: Angle Z shifted after rotating many times ,although X & Y were perfect ?

Elkhawaga, 2019-05-30:
Angle Z shifted after rotating many times by -40 or 20 or -11 ,... 
but  X &  Y angles didn't effected even when moving fast 

you are using coff of Gyro  & Acc in Angle X ,Angle Y 
but in Z angle you didn't made that , Why?

angleX = (gyroCoef * (angleX + gyroX * interval)) + (accCoef * angleAccX);
  angleY = (gyroCoef * (angleY + gyroY * interval)) + (accCoef * angleAccY);
  angleZ = angleGyroZ;

thanks alot for your wonderful effort  :)

tockn, 2019-06-03:
Because an accelerometer cannot calculate angleZ, so I use only gyro sensor in calculating angleZ.
If you want to more accurate angleZ, you can use other sensor together. (ex: geomagnetic sensor)

nhphuong91, 2019-07-25:
You can find detailed explanation as well as many more useful information here:
[https://www.w3.org/TR/motion-sensors/#relative-orientation-sensor](url)