tockn/MPU6050_tockn #34: Different calibration factor : powering VS. resetting

Titibo26, 2019-06-03:
Hi there, 
I'm using esp-wroom32 and i just noticed that when we turn the power ON (from usb or from a battery) calibration coef are 2x higher than what they should be. measurment are also 2 times higher.
When i reset the board via the reset bouton, calibration coef are goods, 2 times less than before. Measurment are perfect.

This behavior is hapening every time. 

exemple : 
when i turn it ON it's : (coefX = 4 / coefY = 2 / coefZ = 2)
Each time i reset it's : (coefX = 2 / coefY = 1 / coefZ = 1)

I can bypass this problem easily but i'd like to know why it's happening so i could not program 2 different behavior...

Any idea ?
Thanks

VHD-BGH, 2019-11-06:
i have this prob too!

Titibo26, 2019-11-07:
Did you dig in to know what happen ? I still don't have a solution

VHD-BGH, 2019-11-07:
No I'm not finde a solution yet.

ABartholsen, 2020-03-25:
Anyone solve this?

crowcasso, 2020-09-15:
Ran into an issue that is similar (possibly the same) as what is mentioned above. We are using a GY-521 module that contains a MPU6050 chip.  We noticed two different results based on three ways we start:

#1 -- powered by battery (no Arduino reset) 
#2 -- powered by battery (with Arduino reset after turning on)
#3 -- powered by battery (Arduino connected to computer by USB cable)

The MPU6050_tockn library works as expected for #2 and #3.  For #1 the angle is reported 2x too large. 

Software Fix:
  -- changed MPU6050_tockn.ccp:19 to     writeMPU6050(MPU6050_GYRO_CONFIG, 0x00);
  -- changed MPU6050_tockn.cpp:119 to    gyroX = ((float)rawGyroX) / 131;
  -- changed MPU6050_tockn.cpp:120 to    gyroY = ((float)rawGyroY) / 131;
  -- changed MPU6050_tockn.cpp:121 to    gyroZ = ((float)rawGyroZ) / 131;

Notes:
It seems as if the gyro configuration sometimes fails (hardware issue?). At least for startup #1 it did for us.  The library currently is trying to configure the gyro to run at +-500 degrees/second (0x08 -- raw value divided by 65.5). If that fails to be configured properly (or fails), it looks like the IMU is defaulting back to running at +-250 degrees/second (0x00 - raw value divided by 131). So, we just force it into the slower method, which at least makes the IMU report angle consistently for all three start methods we tested.

Titibo26, 2020-09-16:
> Ran into an issue that is similar (possibly the same) as what is mentioned above. We are using a GY-521 module that contains a MPU6050 chip. We noticed two different results based on three ways we start:
> 
> 1 -- powered by battery (no Arduino reset)
> 2 -- powered by battery (with Arduino reset after turning on)
> 3 -- powered by battery (Arduino connected to computer by USB cable)
> 
> The MPU6050_tockn library works as expected for 2 and 3. For 1 the angle is reported 2x too large.
> 
> Software Fix:
> -- changed MPU6050_tockn.ccp:19 to writeMPU6050(MPU6050_GYRO_CONFIG, 0x00);
> -- changed MPU6050_tockn.cpp:119 to gyroX = ((float)rawGyroX) / 131;
> -- changed MPU6050_tockn.cpp:120 to gyroY = ((float)rawGyroY) / 131;
> -- changed MPU6050_tockn.cpp:121 to gyroZ = ((float)rawGyroZ) / 131;
> 
> Notes:
> It seems as if the gyro configuration sometimes fails (hardware issue?). At least for startup 1 it did for us. The library currently is trying to configure the gyro to run at +-500 degrees/second (0x08 -- raw value divided by 65.5). If that fails to be configured properly (or fails), it looks like the IMU is defaulting back to running at +-250 degrees/second (0x00 - raw value divided by 131). So, we just force it into the slower method, which at least makes the IMU report angle consistently for all three start methods we tested.

Nice to know, i'll check that.
My fix was simply to call twice time mpu.begin() and since then, it works like a charm.

IanEThompson, 2021-08-08:
Thankyou, thankyou THANKYOU!
I was going nuts trying to work out why my robot as going at 45 degrees when I programmed it to go at 90 degrees, 
When I tried to debug it with the serial monitor it would go 90 degrees, but without the serial monitor, it went 45 degrees.
I've simply used @Titibo26 's fix of 'beginning' twice. Works perfectly now.
Thanks again!