tockn/MPU6050_tockn #35: accZ is always -2.00 on ESP32

TheAnosmic, 2019-07-31:
I'm using mpu6050.calcGyroOffsets(); on setup while it's resting on my table.
The acceleration I get (with the example code only) are:
accX : -1.50    accY : -1.19    accZ : -2.00
X and Y are weird because it's sitting still on my table, but they at least change.
the Z stays at -2.00 no matter how I wiggle it.
it's a gy521 connected with I2C to ESP32 dev board.
any ideas on how to debug it?

Titibo26, 2019-08-01:
Did you check your input voltage ?
Did you check if the mpu is well seen with I2C scanner ?

If yes, did you try to reset the esp ? Each time im using the mpu, the first time im turning on the mpu it's giving me wrong datas but when i reset it, it's working correctly. I didn't find why it's doing that (probably some registers inside the esp or the mpu wich are not going on the same state when turning up and reseting...)

TheAnosmic, 2019-08-01:
Just checked with a different 5V supply just for the MPU, it's the same.
the I2C communication works, I see all of the other information flow as it supposed to, it's just the accZ.
I'm resetting everything all the time, that's part of the project I'm working on.

I will try to get another gy521 to test it.

pjphel, 2020-01-29:
I think its supposed to be 3.3V not 5V

Titibo26, 2020-05-28:
Even in 3.3V it is doing the same :(