tockn/MPU6050_tockn #41: Missing Yaw transfer

Pascal-Lohscheidt, 2020-01-30:
Hey, I just wanted to use your code for my esp8266, and I realised after looking at it that there is not Yaw Transfer ongoing. As far as I know, if you pitch your device and then yaw it the pitch will be not correct anymore.
You can see why in this video clip. I´ve already added a time mark.
https://youtu.be/4BoIE8YQwM8?t=606

Maybe I just was blind and didn´t see it and you´ve already added it.  
Maybe if have the time I can change it myself

bart, 2020-04-03:
Yes, and unfortunately it also looks like the yaw angle gets wrong after moving the sensor around for a few seconds in each axis. When moving back to start orientation pitch and roll are about 0 again but yaw is about 45 degrees off. Might it be the same reason? Any updates on this @Pascal-Lohscheidt ?

Thanks!

Pascal-Lohscheidt, 2020-04-03:
Unfortunately, I didn't have much time in the last month's so I had to
pause the project. But I'm sure it's probably just a few lines of code.
The video explains it pretty good.

bart, 2020-04-03:
So you think the reason for wrong yaw angle over time is the missing yaw transfer, too?

Pascal-Lohscheidt, 2020-04-03:
Well I can't prove it as I have no idea of your set-up.  But I can imagine
that is a valid reason. Another reason could be your loop frequency. If
it's low it could cause a lot of measure gaps.

bart, 2020-04-03:
I used the default get angle example which gives me this issue/ So seems to be a general problem caused by library. I will try to build an own function that generates angle by taking raw accelerometer and gyro values into account. Let's see if it works. After implementing this I hopefully know what is missing in tockn library

Pascal-Lohscheidt, 2020-04-03:
Yeah, I did this before and switched to that library because of issues
after switching from arduino mega to esp

bart, 2020-04-03:
I think I'll still use the library to get raw values but but not the angles. I'll let you know.

carlosgellida, 2021-07-22:
This library do not gives the Euler angle, it only integrates the angular velocity, which isn't the same. I'll change to other library...