tockn/MPU6050_tockn #43: Measurement units of raw data of acceleration and angular velocity

GrigoriyAY, 2020-04-07:
Could you tell, what is the measurement units of raw data of acceleration and angular velocity?
I mean variables here:
`rawAccX = wire->read() << 8 | wire->read();`
`rawAccY = wire->read() << 8 | wire->read();`
`rawAccZ = wire->read() << 8 | wire->read();`
`rawGyroX = wire->read() << 8 | wire->read();`
`rawGyroY = wire->read() << 8 | wire->read();`
`rawGyroZ = wire->read() << 8 | wire->read();`

bhupiister, 2020-08-10:
> @tockn, could you tell, what is the measurement units of raw data of acceleration and angular velocity?
> I mean variables here:
> `rawAccX = wire->read() << 8 | wire->read();`
> `rawAccY = wire->read() << 8 | wire->read();`
> `rawAccZ = wire->read() << 8 | wire->read();`
> `rawGyroX = wire->read() << 8 | wire->read();`
> `rawGyroY = wire->read() << 8 | wire->read();`
> `rawGyroZ = wire->read() << 8 | wire->read();`

Did you get the solution ?

rfetick, 2020-11-11:
Raw data is given in units of bits. The conversion to physical units of m/s² or °/s depends on the calibration of the MPU6050. You can have a look at the [Invensense MPU6050 documentation](https://invensense.tdk.com/products/motion-tracking/6-axis/mpu-6050/). Different calibrations allow to modify the full range and sensitivity of the device.

Y-Grigoriy, 2021-03-31:
@rfetick thank you for clarification. You are right.

closed 2021-03-31 by Y-Grigoriy