tockn/MPU6050_tockn #50: Value by getangleZ() keeps on increasing with time. Even the sensor is still.

MuhamdShahab, 2023-02-12:
Hello there,
I have tested the library for MPU6050 on three different sensors. I have faced a common issue where the angle given by getangleZ() kept on increasing after decimals which eventually generates a bigger difference. This is a kind of cumulative error.

netpipe, 2023-02-12:
drift because of gravity likely there is a mode for extra precision with a gravity factor to fix that iirc.

MuhamdShahab, 2023-02-12:
For sure gravity val effects the precision  but the noise is cumulative. In case of drift there must b a random behaiviour.

MuhamdShahab, 2023-03-05:
@tockn ??

MrXu-98, 2023-10-16:
Hello, has this problem been solved? I found another problem similar to this problem. When x axis and Y axis obtain a certain Angle value, irregular jump will occur, and the Angle value can reach 600+

closed 2023-10-16 by MuhamdShahab